CR6Community/Marlin #336: [BUG]  reaching target temperature does not trigger action (levelling or feeding)

Raekon, 2023-02-06:
### Did you test the latest `extui` code or prerelease?

No, there is no prerelease

### Bug Description

I just installed 6.1 firmware on a CR-6 SE with a 4.5.3 motherboard.

Both firmware updates goes through nicely. 

When I try to level, print, feed filament, anything that requires an action to be triggered by reaching a target temperature, nothing happens. 
The nozzle and bed temp just keep fluctuating around the target temperature (so the sensors are working)

I have tried resetting it to factory defaults and restarting and changing target temps.  nothing. 
And since it is heating, I can't get out of the process without turning the printer off. 

as a side note, when I try to enter a lower value on the "feed"-screen, it just reverts to 150 automatically. 

Homing works ok. 

I tried reverting to Creality Firmware, and everything works again. 

My serial number is P202K03D01...  and since J comes after J, I assume that it is a 4.5.3 motherboard. Also the version number on "info" was later than the numbers mentioned in the "edge of tech" video.

I dont know how to "get" the configuration files... I havent modified any files of those names...

Thanks

### Bug Timeline

happened since upgrading firmware. Not in stock firmware

### Expected behavior

I expected it to start levelling after reaching the target temperature. 

### Actual behavior

It did not move.  just kept rasing and lowering the temps around the target temps.

### Steps to Reproduce

turn on printer.
choose Auto levelling

or

turn on
choose "feed" wait for the nozzle to heat
push "feed again" 

### Version of CR6Community Firmware

6.1 latest release 

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.3

### Display

Creality CR-6 stock touch screen

### Mods

none

### Add-ons

none

### Bed Leveling

ABL Linear grid

### Your Slicer

Cura

### Host Software

None

ritchiedc, 2023-02-06:
Did you run the PID calibration?

Raekon, 2023-02-06:
No, I did not. I tried to print the calibration print, but it wouldn't until I had leveled (which I can't.)I figured, that went for the rest of the options as well... I will try it ASAP (It's a work printer, so not today)Sendt fra Outlook til Android

ritchiedc, 2023-02-06:
Some printers, but not all, do not work with the CF default PID values. Whenever I factory reset I always do a PID calibration. I should note that only the nozzle PID is run from the screen UI. But I've never had problems with the bed PID so I don't go to the trouble of running that from gcode.

Raekon, 2023-02-06:
Okay, thanks.
I'll post an update, when I have one.

Thinkersbluff, 2023-02-06:
The firmware waits for temperature to stabilize at target, before triggering the next action.  When temperature is varying +/- a few degrees around target, that can be enough to prevent Marlin from passing that decision point.

PID with fan on should fix that, if you also have the silicone sock on the heater block.