CR6Community/Marlin #340: Display turns black after Flashing V6.1

Grillfreund, 2023-03-20:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

I tried to flash Version V6.1 on my CR-6 SE (Board Version CR-ERA_V1.1.0.3). Touchscreen Flash was sucessfully but the printer did not accept the 6.1 firmware.bin

So I tried to go down to version 6. After Flashing the touch screen again, the touch screen turns now black (backlight on). If I try to flash again another touch screen firmware, it also stays black. Only if I place an micro SD card with an empty DWIN_SET folder, I see that I currently run DGUS2 V3.5 on my display. But as long as I have files in my DWIN_SET folder, the display turns black.

I also tried to expeiment with the touch screen refactored version. But without success. Can somebody help?

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v1.1.0.0-ERA

### Display

Creality CR-6 stock touch screen

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

ritchiedc, 2023-03-20:
Do you know the original version of DGUS2 OS?

How old is your printer?

I suggest you post your issue on the community discord server. You may get quicker help.

Thinkersbluff, 2023-03-21:
When does the screen turn black?  
- When it starts the flashing process, or 
- after you take out the card and power up?

Grillfreund, 2023-03-21:
Hello all,

thank you for your replys.

@ritchiedc
the printer is now around 1 year old. Unfortunately I don't know the original DGUS2 OS Version but I remember there was once a first text line in red on the display. And if I understand correct it means I updated accidently something. At this point it was not clear for me what I did.

@Thinkersbluff 
Both. If I turn the printer on with flash files on SD card. It shortly flashes blue and turns immediately black. If I turn it on without SD-Card it stays black.
Only if I turnit on with an SD card with an empty DWIN_SET folder on it it stays blue with info that 0 files where copied.

Thank you for any help...

Thinkersbluff, 2023-03-21:
Since you remember one flash showing a red line at the top of the screen and you are now running DGUS2 v3.5, it sounds like your display did not used to run DGUS2 v3.5. 

By itself, that does not explain the black display, but if you do solve this problem and then find the touchscreen does not respond to menu selections, you may still 
need to revisit this point. 

I have noticed that the flashing process normally starts with a black screen, then switches to blue, then starts listing the various actions it takes.   It sounds like your black screw is something else. I will think on this a little.

ritchiedc, 2023-03-21:
If you didn't specifically include the DGUS2 v3.5 files when you tried flashing release 6.1 then it is likely you saw the read text when you flashed release 6 to the screen. The DGUS OS files were included in the DWIN_SET folder prior to release 6.1.

When you originally flashed the screen did you use the files from release 6.1 or did you use the files from "_CR-6_new-hardware-revision-touchscreen-firmware61F_220622_v1.zip"

Thinkersbluff, 2023-03-21:
If it is the display bootloader that is turning the screen black, it may be having trouble reading the SD card boot record.  I have had that happen to me.  I once ordered a replacement screen because multiple cards gave me the same issue. One day a new card just worked. (Go figure!)
If you know how to use the Windows Disk Manager, or if you have a Partion Manager software, you might try removing all partitions from the card and starting again with one MBR Primary Partition of 8Gb or less, formatted FAT32, 4096 allocation units per sector.

Grillfreund, 2023-03-22:
I try to get today another micro SD Card. Today I have 2 hat Home. I'll give feedback as soon as possible...

In between I tired also to just flash the T5L...CFG. This was possible. I had an blue screen and I saw on my display, that one CFG file was flashed. After this I tried again to flash the other files (without CFG file). But after inserting SD Card and starting the printer I had the same issue as before. Screen flashs once in blue and then it turns black.

Thinkersbluff, 2023-03-22:
That is interesting.
You can flash each of the .bin and each of the ICL files, one by one, until you have flashed them all.  Maybe one particular file is causing the problem.
If you can identify one or more problem file, please let us know which one(s) it is and try downloading, extracting it/them again, in case it/they got corrupted during that process.  If the problem is the card itself, one by one might actually work…

Grillfreund, 2023-03-24:
Hello @Thinkersbluff ,

I still wait for another micro SD card. But in between I tried to flash with my current SD the files one by one as you proposed in your last post. 

The behaviour is strange. I'm only able to flash the CFG file correctly. If I try to flash any other file, the display turns black. My last chance is to test another SD card. If this will also not work, I order a new display for my printer :-(

I keep you updated.

Thinkersbluff, 2023-03-24:
> 
> …I tried to flash with my current SD the files one by one as you proposed in your last post.
>…I'm only able to flash the CFG file correctly. If I try to flash any other file, the display turns black…
> 

I am sorry and a bit mystified to hear that only the CFG file will flash correctly.  Thank you for trying and for letting us know the result.  I hope the new card does the trick.

Grillfreund, 2023-05-25:
Unfortunately not. 
I ordered a new display.

Michael

Thinkersbluff, 2023-05-25:
Sorry to hear that.
Apparently, it is possible to corrupt the bootloader on the display, if the flashing process is interrupted or if the configuration file is misread (or an incorrect configuration file is flashed).  DWIN now sells a T5L "burner", and [Maria Guo (on YouTube) has offered to provide whatever files have to be reflashed to the display to restore one, once it has gone black as a result of this issue. ](https://www.youtube.com/watch?v=Ij_FfQagPvg) The problem here, I think, is that it is not a stock DWIN display, but rather a T5L-based custom-built & custom-coded display designed & assembled by Creality themselves, so Maria may not have the necessary files & Creality may not be "sharing".

Replacing the display is extreme, but likely the only available option, in cases like yours.
Fingers crossed that the next one works for you.

closed 2023-05-25 by Thinkersbluff