CR6Community/Marlin #352: [BUG] (enter short bug summary here)

MathieuMH, 2025-01-17:
### Did you test the latest `extui` code or prerelease?

No, I cannot complete

### Bug Description

Hi all,

I have my CR-6 SE for quite some time. And it was little abandonned. So decided to remove the dust and install my PEI build plate, aligned everything again properly, update to latest touchscreen firmware (1.1 refactored), and latest printer firmware (6.1 community).
Then I took it for a spin, started with bed leveling where it immediately went sideways.

*Problem*
Auto bed leveling fails after the first point or 2nd point. Homing works flawless, the center position is also without problem.
It seems that it's trying to go too deep, I hear the mechanics making unusual noises after it reached the bed.

*Attempts*
1. Increase z-offset, although I don't think it has much impact, but I increased it up to almost +1mm, and still the problem occurs.
2. Decrese z-offset, as above, this seems to make no difference.
3. Factory reset, to be sure I don't have any exotic settings floating around I tried the factory default, same result.
4. Check the piezo element, (responsible for detecting the pressure on the hot-end), that seems to work fine, I see the LED going on when I briefly apply pressure on the nozzle of the hot-end.
5. Check he Z-axis home sensor, also there works properly...

I'm running out of idea's. 
I suspect it to be a SW bug, as before (without latest touch and printer firmware) this wasn't occuring at all.

Any suggestions on further identifying the cause?

Kind regards,
Mathieu Hebbrecht

### Expected behavior

Ability to complete auto bed leveling.

### Actual behavior

Fails always after first or second step.
It seems it is trying to go to far into the bed. Mechanics of the printer don't like that.

### Steps to Reproduce

1. start printer
2. go to auto bed leveling
3. start it
4. first is home's the printer (no issue, also to too deep into the bed driving here)
5. once temperature is reached it starts the mesh
6. First or 2nd step fails, printer is stuck forever in the bed leveling
7. power cycle the printer to escape.

### Version of CR6Community Firmware

CR-6 SE v4.5.2 board (6.1 FW with 1.1 refactored touchscreen FW)

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

None

### Add-ons

None

### Bed Leveling

None

### Your Slicer

Cura

### Host Software

SD Card (headless)

ritchiedc, 2025-01-17:
It's likely that the tab that blocks the optical sensor is hitting the top of the sensor before the nozzle touches the plate enough to trigger the pressure sensor. This happens with some printers when switching to a pei bed because it can be slightly thinner then the glass plate.

Thinkersbluff, 2025-01-17:
“ First or 2nd step fails,”

Please clarify what you mean by the first and second “step”

-  Does the printer sometimes complete probing the first point, then fail on the second?  

-  Does the probe reach the bed, or is the target tab striking the top of the optical sensor, as suggested by @ritchiedc ?

MathieuMH, 2025-01-18:
Thanks for the reactions.
@ritchiedc I'll monitor the Z-axis home sensor. Did not pay too much attention to it before. In case this is the cause, what is the suggested solution. Extending the tab, or move up the sensor?

@Thinkersbluff as you suspected, the first one succeeded as normal, then 2nd failed. Sometimes it even failed at the fist, since I played a bit with the Z-offset it reaches the 2nd point always.

ritchiedc, 2025-01-18:
There is an adapter someone designed but you need a working printer. I took a drill to the sensor mounting screw holes and made them oblong so I could move the sensor down a millimeter or so.

Thinkersbluff, 2025-01-19:
Sounds like this is not a software bug.

closed 2025-01-19 by Thinkersbluff