CR6Community/Marlin #353: [BUG] Can't compile firmware for Creality V4.5.2

Byrnes9, 2025-03-11:
### Did you test the latest `extui` code or prerelease?

Yes, and the problem still exists.

### Bug Description

Trying to compile firmware, getting error:

Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

### Bug Timeline

n/a

### Expected behavior

Firmware to compile successfully

### Actual behavior

Firmware is failing with this error:
Warning! `src_filter` configuration option in section [env:DUE] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:esp32] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_build_flags` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_flags` instead
Warning! `src_filter` configuration option in section [env:linux_native] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:SAMD51_grandcentral_m4] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:malyan_M300] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy31] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy35] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy36] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Warning! `src_filter` configuration option in section [env:teensy41] is deprecated and will be removed in the next release! Please use `build_src_filter` instead
Processing STM32F103RET6_creality (platform: ststm32@~12.1; board: genericSTM32F103RE; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------MissingPackageManifestError: Could not find one of 'package.json' manifest files in the package

 *  The terminal process "C:\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

[Marlin-2.0.8.1-cr6-community-release-6.1.zip](https://github.com/user-attachments/files/19189690/Marlin-2.0.8.1-cr6-community-release-6.1.zip)

### Steps to Reproduce

1. Open VScode
2. Put creality board into platformio.ini and configuration
3. try to compile

### Version of CR6Community Firmware

6.1

### Printer model

Creality CR-6 SE

### Motherboard

Creality v4.5.2

### Display

Creality CR-6 stock touch screen

### Mods

Microswiss hotend

### Bed Leveling

ABL Bilinear mesh

### Your Slicer

Prusa Slicer

### Host Software

None

lifendel, 2025-03-11:
This is a known issue as there were changes in some of the syntax for compilation that prevents the older Marlin code from compiling with the latest platformIO. To fix this, you must do the following:

You edit marlin.py in the following folder:

buildroot/share/PlatformIO/scripts/marlin.py

Go to line 19 for the following:

def replace_define(field, value):
    for define in env['CPPDEFINES']:
        if define[0] == field:

Edit and replace the second line with the two lines as shown below:

def replace_define(field, value):
    envdefs = env['CPPDEFINES'].copy()
    for define in envdefs:
        if define[0] == field:

Save that and you can now compile the CF6.1 source.

Byrnes9, 2025-03-11:
I did replace the Marlin.py file with the recommended one.

I was able to get it to compile by deleting the .platformio folder in my C drive and reinstalling the extension.

Thank you for your help though.

Byrnes9, 2025-03-11:
Another issue has come up

When I select reset factory settings, the screen freezes and I have to hard reset the machine, I haven't changed the files beyond replacing the Marlin.py and setting my motherboard in platform io.

lifendel, 2025-03-11:
Which screen firmware do you have? You should have the firmware bundled from one of the 6.1 firmwares or the re-factored firmware for it to be compatible. What are you changing in the Configuration.h?

Byrnes9, 2025-03-12:
I'm currently using the refactored screen firmware. I haven't changed anything in configuration.h other than using the 4.5.2 config provided in the source code file.

lifendel, 2025-03-13:
Please be aware that enabling the heater during leveling and ABL probes will reducing probing accuracy. It was found during testing that the current switching on an off to maintain the heater temp was causing fluctuations in the readings from the strain gauge. Also, you can swap two pins on the JST connector for the stepper motor to reverse the direction, then you do not need to modify the firmware. That being said, I do not know why that would cause issues with the factory reset locking the system. I'm curious if it is just with the screen or the mainboard itself. Can you attach to the USB and still issue commands to the mainboard via a terminal software like Pronterface? Or does the mainboard go dead when issuing a factory reset from the screen? From the terminal you can also clear the EEPROM by issuing a M502 followed by a M500. M502 resets everything to factory defaults and M500 then writes those changes to the EEPROM.

Byrnes9, 2025-03-13:
It may have been a motherboard issue, I plugged it in and the printer shorted. Gonna have to get a new motherboard. SD card also shorted but my computer is fine.

lifendel, 2025-03-13:
Then you may have the 24VDC Short to ground on the bed. If you have a 4.5.2 mainboard, then like me, you have an original kickstarter unit. It was found that the bed clamps for the glass are bolted over the heater traces for the bed. Those clamps can pierce through the insulation on the traces and short the traces to ground. That will cause several issues when plugging into the USB as it will send 24VDC down the USB ground. There was a 2nd gen bed that was made which re-routed the traces, but for the original bed, the fix was to place Kapton tape between the bed and the clamps. Attached is a picture of the two beds. The right is the V2.0 bed, the left is the V1.0 bed with Kapton tape where the bed clamps bolt on.

[image]

Byrnes9, 2025-03-13:
Thank you for your help. Can I replace remove the clamps without uninstalling the bed? I have a flexible build sheet on it and with the price to replace that and the motherboard, I may as well just replace the machine.

lifendel, 2025-03-13:
Yes, you can just remove them and leave them off as I did when I put on the flex buildplate. There's no reason to replace the bed if you remove or insulate them with Kapton tape.

Byrnes9, 2025-03-13:
Thanks, sorry I meant do I need to remove the bed from the machine to remove the clamps or can I remove the clamps without removing the bed and the flex buildplate magnetic base?

lifendel, 2025-03-14:
If you did not cut holes for the screws in the magnetic before applying, you can use an exacto knife to cut a hole to access the screwhead. I don't recall if they have nuts on the bottom or not as I took the clamps off my machine as soon as I got it as I immediately got a textured PEI bed for it when I kickstartered it.

Byrnes9, 2025-03-14:
I think the back clamps have nuts and the front don't. Thank you for  all your help, I'll update on whether I went down this option or not.

Byrnes9, 2025-03-14:
Update: I have decided to dispose of the printer. Thank you for all your help with it.

closed 2025-03-14 by Byrnes9